adafruit/Adafruit_FeatherOLED #1: Incorrect Endian Representation in renderIPAdress

alexanderlin96, 2016-06-17:
- Arduino board:  Adafruit Feather HUZZAH ESP8266

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.69

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):

```
#include <WiFiClient.h>
#include <ESP8266WiFi.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <Adafruit_FeatherOLED.h>
#include <Adafruit_FeatherOLED_WiFi.h>
Adafruit_FeatherOLED_WiFi oled = Adafruit_FeatherOLED_WiFi();
void setup(){
    oled.setIPAddress((uint32_t)WiFi.localIP());
    oled.refreshIcons();
}
void loop(){
}
```
On the board we see the IP address reversed. I am guessing this has something to do with the class IPAddress representing IP in Big-Endian, however I believe the Arduino and Adafruit Feather HUZZAH ESP8266 are Little-Endian. In Adafruit_FeatherOLED_WiFi.cpp:
```
void Adafruit_FeatherOLED_WiFi::renderIPAddress ( void )
{
  if (_ipAddressVisible)
  {
    if (_connected)
    {
      setCursor(0,24);
      print((_ipAddress >> 24) & 0xFF, DEC);
      print(".");
      print((_ipAddress >> 16) & 0xFF, DEC);
      print(".");
      print((_ipAddress >> 8) & 0xFF, DEC);
      print(".");
      print(_ipAddress & 0xFF, DEC);
    }
  }
}
```
the above is the originally library code which is the correct representation for Big-Endian, but not for Little-Endian. I changed this portion of the code in my local library to:
```
void Adafruit_FeatherOLED_WiFi::renderIPAddress ( void )
{
  if (_ipAddressVisible)
  {
    if (_connected)
    {
      setCursor(0,24);
      print(_ipAddress & 0xFF, DEC);
      print(".");
      print((_ipAddress >> 8) & 0xFF, DEC);
      print(".");
      print((_ipAddress >> 16) & 0xFF, DEC);
      print(".");
      print((_ipAddress >> 24) & 0xFF, DEC);
    }
  }
}
```
and now the IP displays correctly on my OLED display.

projectarc-co-uk, 2016-12-14:
Confirmed - this change worked for me too.

Using an Adafruit Huzzah Feather with the OLED Featherwing the IP address obtained from WiFi.localIP() (via ESP8266WiFi.h) displayed in reverse order when passed to setIPAddress(ip). 

Changing Adafruit_FeatherOLED_WiFi.cpp as described by alin96 meant the IP addresses displayed correctly.

Sembazuru, 2017-02-21:
I can confirm the issue is also there an the ARM platform, specifically [Adafruit Feather M0 WiFi - ATSAMD21 + ATWINC1500](https://www.adafruit.com/products/3010) using WiFi101 library.

I applied the correction above to my local library and can confirm it works for my platform as well.

closed 2017-04-02 by microbuilder