adafruit/Adafruit_FeatherOLED #10: Battery Voltage Divider Pin for 32u4 Feather

braychristopher, 2019-06-18:
In the example code for the battery indicator (for the 32u4 Feather), it tries to read the battery voltage off of the A7 pin. It is actually the A9 pin.

Here is a link to the Adafruit forum post where the Adafruit_Support_Carter helped me figure this out.

Thanks for developing these libraries and code examples.

caternuson, 2019-06-18:
Related:
https://forums.adafruit.com/viewtopic.php?f=57&t=153038

ladyada, 2019-06-18:
thanx - this library is a little crusty - @makermelissa will be taking a sweep at it!

makermelissa, 2019-06-18:
Thanks for reporting this. The A7 input is fine for M0, but A9 was needed for 32u4, so I made a separate `#ifel` block. Referenced schematics to confirm.

closed 2019-06-18 by makermelissa