adafruit/Adafruit_FeatherOLED #21: Adafruit 128x64 OLED FeatherWing Support?

Andrewiski, 2021-02-17:
- Arduino board:  **ESP32 FeatherWing**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.12**

Any chance this library is scheduled to be updated to include the larger Adafruit 128x64 OLED FeatherWing?  

The awesome battery and Wifi icons work so well with the smaller OLED display leaves my larger display lacking ;(

ladyada, 2021-02-17:
no current ETA but...If you submit a PR to add support we'd add it :)

Andrewiski, 2021-02-18:
It looked like the current version is including the other ole file directly so duplicated the files and added SH110X. Changes were with setRotation(1),  WHITE -> SH110X.  Some swap arounds with width and height.   I am using an esp32 so added that to the two battery examples as well since I was already in there.  

Thanks for all you do.

Andrewiski, 2021-02-18:
Not sure what the error is but I do know Clang never loves me

Andrewiski, 2021-02-18:
I thought last time I had a clang issue it was a line that was to long but I am not able to figure out what it doesn't like.  It just says error.

Andrewiski, 2021-02-18:
Tweaked the battery voltage calc and finally got Clang and Dyoxon to give me some love.

ladyada, 2021-02-18:
good work :)

closed 2021-02-18 by ladyada