adafruit/Adafruit_FeatherOLED #29: Wrong formula for voltage divider

axelmagnus, 2022-03-10:
- Arduino board:  **Huzzah ESP8266**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.16** (irrelevant)

The formula for the divider is wrong:

`#elif defined(ESP8266)
// esp8266 feather
#define VBATPIN A0
/*
 * The ESP8266 Feather Huzzah does not have an internal voltage divider, so you
 * must build one. However, the voltage on the Analog Input must be between
 * 0-1.0V or you could risk damaging the chip. So please keep this in mind when
 * selecting resistor values. For the default values you could use a 100K
 * resistor between ADC and USB (R1) and a 10K Resistor between ADC and Ground
 * (R2) to use these values.
 */
#define VBAT_MULTIPLIER (100000 / 10000) / 1024.0`

**It should be** 

`#define VBAT_MULTIPLIER (110000 / 10000) / 1024.0`

but I would rather have code where you can change the resistors:

`
#define R1 100000
#define R2 10000
#define VBAT_MULTIPLIER ((R1+R2) / R2) / 1024.0`

Thank you!

ladyada, 2022-03-10:
@caternuson did ya just fix this?

caternuson, 2022-03-10:
The recent PR #28 removed redundant math to use just `VBAT_MULTIPLIER` to do the conversion, so that battery voltage is computed simply as:
```cpp
analogRead(VBATPIN) * VBAT_MULTIPLIER;
```
But it do not change any of the actual `VBAT_MULTIPLIER` definitions.

Looks the issue here is the minor voltage divider math in the #def for ESP8266?

caternuson, 2022-03-10:
@axelmagnus Want to PR something?

axelmagnus, 2022-03-10:
My 1st PR. did i do it right, @caternuson ?

axelmagnus, 2022-03-15:
Solved with #30

closed 2022-03-15 by axelmagnus